tfabris/BlueGigaEmpeg #69: Intermittent crackling static on I2S developed recently.

tfabris, 2018-02-02:
Yesterday the I2S audio developed an occasional intermittent crackle sound. Like a digital version of dust on an LP record. This is after weeks of perfect operation.

I have checked all voltages and done a ton of diagnosis, including ruling out the Empeg, the wiring, the Arduino, and my new Rev 6 board. My current theory is that either the chip or my current Betz board is going bad.

This is troubling because I've fried chips before, and I thought I had solved all those problems. Another Betz board is on its way, and if that one doesn't crackle then I'll have to think about whether I need to look at power issues AGAIN.

tfabris, 2018-02-03:
Tried all of these, did NOT fix:
- Different Empeg
- Different Betz board and WT32i chip
- Different Arduino
- Different Bluetooth output devices (Honda, Plantronics, etc.)
- Different wiring harness
- Fiddling with wiring proximity and twisting of I2S wires on wiring harness
- Different resistor values of voltage dividers (lowering voltage to below 2.5v by two different amounts)
- Various changes to the I2S parameters in the BlueGiga code (removing keepalive, changing bit depth value, changing event parameter, etc.
- Disconnecting serial port
- Removing power from the MAX232 device
- Removing the Pololu power supply and running the Arduino off of 12v.

Tried this and it SEEMED to fix the problem briefly at first but then later, tried again and got the crackling back again:
- Different BlueGigaEmpeg PCB (rev3 with handmade voltage divs versus Rev6 with built in divs)

What does seem to fix the crackling:
- Switching to analog audio. Though I didn't try it with actual sound on the analog channel (didn't have the right cable connector) so I should try that again with actual sound on the analog audio channel. In any case, it doesn't help really, but I think it confirms that the issue is in the I2S and not in the Bluetooth.

TO DO:
- Try lowered voltage pololu (try back to 5v or 6v maybe)
- Try yet another different Betz board
- Try analog audio except this time actually pump some audio through that connection
- Try changing routing of I2S audio wires on the BlueGigaEmpeg board to look for possible proximity interference

mlord, 2018-02-06:
Could it be a grounding issue?  Eg.. run a wire from GND on the empeg (eg, the chassis) to GND on the Betz board, and see if problem goes away.

tfabris, 2018-02-06:
Thanks @mlord, I will definitely try that! I have high hopes for putting the C16 470uf cap back as well, I'm about to try that right now.

tfabris, 2018-02-06:
Tried the following, unfortunately did not work:
- Adding additional grounding between bluetooth board and the empeg chassis as @mlord suggested.
- Adding additional grounding to the AGND line on the bluetooth board.
- Removing grounding to the AGND and cutting the AGND jumper on the bluetooth board.
- Adding back the 470uf cap at C16.

Tried the following which seemed to maaaaaaybe help:
- Used long standoffs to separate the Betz board from the BlueGigaEmpeg board by about an inch or two. This seemed to reduce the crackling but did not eliminate it.

TO DO:
- Working on the theory that this is some kind of proximity interference, try rerouting the I2S connections on the BlueGigaEmpeg board completely and see if their positioning affects the outcome.
- Working on the theory that adding back the C16 470uf cap was the right idea, but that all my Betz boards now have a damaged internal regulator inside the WT32i chip because I removed the cap, await the arrival of the new set of Betz boards from China and see if leaving the cap in place on a new board solves the problem.

tfabris, 2018-02-06:
Works:
- Handmade test, running the I2S connections and the voltage divider resistors outside/away from the board assembly.

This tells me that the problem is some kind of interference, but what/how. Why did the R3 board have less of it than the R6 board?

Things to think about:
- R3 board had hand-made voltage dividers with slightly more separation between their lines than the R6 one has on its copper traces.
- R3 board had those voltage dividers positioned over two layers of ground plane, and that ground plane was between the vdivs and the Arduino CPU. Perhaps it's Arduino RF emissions tangling with the signal on the I2S lines?
- The I2S connections, the wires themselves, on either the R3 board or my handmade test, might be a bit thicker than the traces on the R6 board. Maybe thicker traces on the next R7 board?

tfabris, 2018-02-06:
Moving the SCK wire closer to the Arduino produces some pops.

tfabris, 2018-02-06:
Moving WS wire closer to the Arduino did not immediately produce any pops.

tfabris, 2018-02-06:
Moving the SD wire closer to the Arduino did not immediately produce any pops.

Theory:
- Only the SCK wire is the issue, and we merely need to route it as far away from the Arduino as we can and make sure the voltage divider resistors are protected by ground planes.

tfabris, 2018-02-06:
Tried:
- Routing the SCK wire differently and putting all voltage divider resistors on the opposite side of the PCB as the Arduino. Also completely cutting the original SCK trace at both ends so that it doesn't act as an antenna. This didn't fix the problem. My guess is that it's because in the Rev6 design there's not a lot of ground plane shielding between the voltage divider resistors and the Arduino.
- Fashioning a makeshift RF shield out of aluminum foil covered with tape and using it to cover the Arduino chip itself. Didn't really seem to do much, but that might be because I didn't ground it well or because it wasn't properly shaped.

Stumped, because the act of printing an entirely new PCB with a theoretical design is a bit of an outlay of time and expense and I'd hate to do all that and then not have it be fixed.

tfabris, 2018-02-06:
Tried:
- Cutting traces for all three I2S lines across the entire BlueGigaEmpeg PCB and, while still leaving the voltage dividers in place, running wires up and over the Betz board. This significantly reduced the number of pops though it did not completely solve it. Promising though.

Consider:
- Possibly a complete redesign of the PCB which rotates the Betz board 180 degrees to place the I2S wires closer to the tuner connector so that I don't have to run I2S traces all the way along the BGE board. Shortest possible I2S runs. Maybe even reposition the RESET/PAIR switch and LED and other elements, to allow the Betz board to get close to the  since I haven't finalized the enclosure design either. This would be a large pain but it might be the only way to be sure.

tfabris, 2018-02-06:
To do:
- Read the heck out of this:  http://www.ti.com/lit/an/szza009/szza009.pdf

mlord, 2018-02-06:
Good work.  Google this:  "isolation routing", and try applying it to the SCLK line (and others if you can).
I think sn00p even posted some nice photos of such on the BBS.

tfabris, 2018-02-06:
Hey @mlord what do you think of this? This document here....
http://www.ti.com/lit/an/szza009/szza009.pdf
... says something interesting:

> "In the case of an I/O pin in the input mode, the capacitance of the unused output transistors transfers noise from both power rails to the pin. The amount of noise is based on the impedance of whatever is connected to the pin. The higher the impedance, the more noise comes out of the microcomputer. That is why unused inputs should be tied to the lowest-impedance rail: ground, by direct short, if possible."

Do you think it would be worth it to try simply grounding all the unused analog and digital I/O pins on the Arduino? Right now I just leave them untouched.

I could also, in the code (i.e., no physical board changes), change all I/O pins from their default setting of "INPUT" to the alternative "INPUT_PULLUP", but I don't know if that would help. I was worried that changing them all to pullups would require more overall power draw for the Arduino board.

tfabris, 2018-02-06:
I'm reading some stuff on the internet such as this...
https://arduino.stackexchange.com/questions/574/should-all-unused-analog-input-ports-be-tied-to-ground-for-accurate-a2d-measurem
... which says that setting the pins to either "INPUT_PULLUP" or setting them to "OUTPUT" and "LOW" (essentially pulling them to ground) will reduce noise emitted by the Arduino. The Pullup uses a small amount of energy, Output/Low uses the least energy. Does that sound right to you, @mlord?

tfabris, 2018-02-07:
To Do:
- Try grounding the RESET line and holding the Arduino in reset mode to see if that stops the interference. That will narrow it down to be the Arduino for sure. Suggested by Shonky on EmpegBBS.
- Try setting all unused pins to Output/Low or INPUT_PULLUP and see if that stops the interference.

mlord, 2018-02-07:
You are not using the A2D (analog to digital sampling) pins, so there is no need to fuss much about those particular pins.  But setting all of the unused ones to INPUT_PULLUP shouldn't hurt -- the internal pull-up is 10's of K-Ohms, so the power draw/waste ought to be pretty small.

tfabris, 2018-02-07:
Tried the following, none fixed the issue:
- Setting all unused pins, both digital and analog, to OUTPUT/LOW.
- Setting all unused pins, both digital and analog, to INPUT_PULLUP
- Holding Arduino RESET line to ground.

I'm not sure if holding the Arduino reset line to ground truly sleeps the chip or not, so more investigation is needed there.

tfabris, 2018-02-07:
Tried, did not fix the issue:
- Putting arduino chip to sleep with this code:

      set_sleep_mode(SLEEP_MODE_PWR_DOWN);
      sleep_enable();
      sleep_cpu();

tfabris, 2018-02-08:
Tried, based on @mlord suggestion:
- Changing 10k voltage dividers out for lower values to increase S/N ratio on the I2S lines at the expense of increased current draw from the empeg.

Result: With 2.33k of resistance (doubled up 4.7k's) there was a significant reduction in pops, though not completely fixed.

TO DO:
- Try 2.0k resistors and redesign the Betz board to rotate it 180 for shorter I2S runs.
- Check empeg interior for hot components in case the additional current draw is a problem.

tfabris, 2018-02-08:
Tried:
- Going down to about 1.55k ohms on the voltage dividers by tripling up the 4.7k resistors (my PCB is a work of art, I tell ya).

Result:
- Pops are almost completely gone. Down to about one pop per song.
- I'm not noticing any unusual heat from the SA7705H or the WT32i.

To do:
- Order a passel of 1.5k resistors.
- Run it this way for a week or so to see if the empeg fries.
- Decide if that's enough, or if I still need a board redesign, too. Keeping in mind that the pops are not noticeable when the music is loud, they are only noticeable in the quiet passages and only if the headunit stereo volume is turned up to a decent amount.

tfabris, 2018-02-09:
Closing issue. 1.55k resistors seems to work and I have 1.5k on the way. Happy dance!

closed 2018-02-09 by tfabris